foxsi/foxsi-4matter #25: Implement downlink with multicast address

thanasipantazides, 2023-07-17:
Need to transmit to a multicast endpoint to use EVTM. So implement multicast UDP send on Formatter. Multicast to EVTM and GSE. Will also need to filter (avoid duplicate) packets on receive in GSE.

yixianz, 2023-09-13:
I don't think duplicate packets would be a problem. I did some tests with one or two laptops - with only one laptop, I just opened different terminal windows for sending and receiving separately; with two laptops in the same local network (e.g. my home Wifi), I could send messages from one to the other. Below are some example scripts that I used and the terminal outputs.
## Simple sender and receiver
Starting from something very basic in `updgarbage.py`, I randomly generate 10 bytes of garbage data every second and send them out by multicasting. Here I'm not mimicking the 18 Mbps downlink rate because I want outputs to be short and clean in my terminal, and I've tested that such rate would work just the same.
```
# udpgarbage.py
import socket, time, random

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM, socket.IPPROTO_UDP)

# multicast group address and port
mcast_grp = '224.1.1.0'
mcast_port = 3000

while True:
	# make some random data to transmit
	data = random.randbytes(10)
	print("sending data to multicast group " + mcast_grp + ":" + str(mcast_port))
	sock.sendto(data, (mcast_grp, mcast_port))
	time.sleep(1) 
```
On the receiver side, `printudp.py` looks like:
```
# printudp.py
import socket, struct

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM, socket.IPPROTO_UDP)
sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)

# multicast group address and port
mcast_grp = '224.1.1.0'
mcast_port = 3000

# bind socket to mcast_grp on mcast_port
sock.bind((mcast_grp, mcast_port))
# convert multicast address to binary form, INADDR_ANY means any interface
mreq = struct.pack("4sl", socket.inet_aton(mcast_grp), socket.INADDR_ANY)
# join the multicast group (adding membership)
sock.setsockopt(socket.IPPROTO_IP, socket.IP_ADD_MEMBERSHIP, mreq)

while True:
	data, sender_endpoint = sock.recvfrom(1500)  #receive data
	print(str(sender_endpoint[0]) + ":" + str(sender_endpoint[1]) + " sent" + data.hex())
```
Testing with one laptop (running udpgarbage.py and printudp.py in two separate terminal windows), I get:
```
% python udpgarbage.py
sending data to multicast group 224.1.1.0:3000
sending data to multicast group 224.1.1.0:3000
sending data to multicast group 224.1.1.0:3000
sending data to multicast group 224.1.1.0:3000
sending data to multicast group 224.1.1.0:3000
```
```
% python printudp.py
192.168.1.149:52535 sent 1d14ac7bd5f1e26db5cc
192.168.1.149:52535 sent 2f09ca6f06c192cf3794
192.168.1.149:52535 sent 80ff21e931dc64c8da68
192.168.1.149:52535 sent 506c10a97c20e56c6ac1
192.168.1.149:52535 sent 740d1a72300e015e1a26
```
Note that in those scripts I don't explicitly write out the standard or loopback IP of my computer - I just tell the sender and receiver to both use the multicast address (i.e. "224.1.1.0") and it works. This is also true when I try multicasting from one computer to another.

## Ways to avoid duplicate packets
### Using different multicast groups/ports
If there are two routes from the sender to the receiver, one way to avoid duplicate packets is to use different multicast addresses and/or ports for each route. Then you can bind your receiver's socket only to the address and port that you want. 

For example, on the sender side, this time I create two multicast groups "224.1.1.0" and "224.1.1.1" both with port 3000. (Using the same group but changing the port would also work.) I send random bytes again through "224.1.1.0" and send "0x01" repeatedly through "224.1.1.1".
```
# udpgarbage2.py (two groups)
import socket, time, random
[... 102 lines cut ...]
```
% python udpredirect.py
Redirecting data from ('192.168.1.149', 59965) to ('224.1.1.0', 3001)
Redirecting data from ('192.168.1.149', 59965) to ('224.1.1.0', 3001)
Redirecting data from ('192.168.1.149', 59965) to ('224.1.1.0', 3001)
Redirecting data from ('192.168.1.149', 59965) to ('224.1.1.0', 3001)
```
Receiver, if  `mcast_port = 3000` in `printudp.py`:
```
% python printudp.py
192.168.1.149:56532 sent 6131211b134ed10b5a0c
192.168.1.149:56532 sent d74a9fac7a0b91105e34
192.168.1.149:56532 sent 2ed95b3dfa530b233945
192.168.1.149:56532 sent c6d3e9708fb54fb3b538
```
If `mcast_port = 3001` in` printudp.py`:
```
% python printudp.py
192.168.1.196:3000 sent 43061d9b3d5f983f8115
192.168.1.196:3000 sent 8cb6ddf754a5be3e91cd
192.168.1.196:3000 sent 017e9cac62c2c99a64de
192.168.1.196:3000 sent e02d9402d8b9dae0524f
```

I didn't get any duplicate packets because the receiver can choose which route it wants to receive data from.

thanasipantazides, 2024-01-24:
Notes to self for implementation: 
- Want to infer endpoint type from `systems.json`.
- In Python, use [`ipaddress.is_multicast()`](https://docs.python.org/3/library/ipaddress.html), then open socket accordingly.
- In C++, use [`boost::asio::ip::address::is_multicast()`](https://www.boost.org/doc/libs/1_84_0/doc/html/boost_asio/reference/ip__address.html), after constructing from `::make_address(std::string address)`.
    - See [this](https://www.boost.org/doc/libs/1_84_0/doc/html/boost_asio/example/cpp11/multicast/receiver.cpp) `asio` example and [this](https://stackoverflow.com/a/12749727) post.

thanasipantazides, 2024-01-29:
In the above `boost::asio` example, sending to a multicast address is as simple as:
1. creating an address with `::make_address()`, supplying a multicast address as `std::string` argument,
2. sending to the endpoint.

This is the exact same workflow as sending to a single endpoint.

It will be a little trickier to get the `Listener` set up correctly in the GSE, but that is in Python and will be quick to debug. Should be possible to determine address type from provided JSON, and conditionally open the correct type of socket and `bind` or `join`.

thanasipantazides, 2024-03-12:
Closing, this was resolved on the Formatter side by edits in foxsi4-commands.

closed 2024-03-12 by thanasipantazides