foxsi/foxsi-4matter #29: Faster build

KriSun95, 2023-08-31:
It takes a lot of minutes to build the executables on Raspberry Pi. For a lot of development, we need to write a little code, then rebuild, then try it out, then write more…. So the long build time slows this sort of prototyping down. The goal of this task is just to speed up the build process on Raspberry Pi. 

We could refactor the project structure [similar to this](https://www.open-std.org/jtc1/sc22/wg21/docs/papers/2018/p1204r0.html), so that instead of all the sources getting built into one library we have multiple that get built separately. Each library would have its own CMakeLists.txt. I would suggest grouping the current source files like this (because these clumps are often modified together):

- commanding
  - Commanding.cpp, Commanding.h, LineInterface.cpp, LineInterface.h
- subsystem
  - Subsystem.cpp and Subsystem.h
- utilities
  - Utilities.cpp, RingBufferInterface.cpp, Fragmenter.cpp, and their headers, and Parameters.h.

I don’t actually know how much this would improve build speed though

*From Thanasi's Google Doc*

KriSun95, 2023-08-31:
This has been started and partially addressed in #28

thanasipantazides, 2023-08-31:
Another option: cross-compile the whole software for the Raspberry Pi architecture. Build done on a faster laptop/desktop machine, then just send the binary over to the Pi to run. There is [a guide on doing this in CMake here](https://cmake.org/cmake/help/book/mastering-cmake/chapter/Cross%20Compiling%20With%20CMake.html).

thanasipantazides, 2023-08-31:
BTW, I edited the starting description to have a different source grouping for the libraries. These are organized by dependency now and may help build faster?

thanasipantazides, 2023-12-19:
Yixian completed! Thanks!

closed 2023-12-19 by thanasipantazides